abs-tudelft/fletcher #61: Integrate Vivado HLS compatible streams into wrapper generation process

johanpel, 2018-10-10:
When user core implementation is done through Vivado HLS it's possible to integrate Fletcher streams with Vivado's stream format (for hls::stream<type> arguments to HLS functions).

This could be done automatically if the user would supply some argument saying they want the stream to be Vivado HLS compatible.

Vivado uses the following signaling for hls::stream. For example on a HLS output top-level we find that an input stream has:

    values_V_V_dout : IN STD_LOGIC_VECTOR (31 downto 0);
    values_V_V_empty_n : IN STD_LOGIC;
    values_V_V_read : OUT STD_LOGIC;

The handshake protocol should be checked but it seems it is assumed the HLS component will pull from a FIFO. Empty_n and read look like valid and ready, but this needs to be checked to be sure.

Best case outcome would be that a user can just instantiate the HLS core in the generated template without having to type a single line of HDL, similar to OpenPOWER SNAP.

johanpel, 2018-10-10:
Example where we just want to add 1 to a column of integers:

```
void increment(hls::stream<int> &number, hls::stream<int>& out) {
	int num;
	number >> num;
	num = num + 1;
	out << num;
}

```
Results in a component:

```
entity increment is
port (
    ap_clk : IN STD_LOGIC;
    ap_rst : IN STD_LOGIC;
    ap_start : IN STD_LOGIC;
    ap_done : OUT STD_LOGIC;
    ap_idle : OUT STD_LOGIC;
    ap_ready : OUT STD_LOGIC;
    number_V_dout : IN STD_LOGIC_VECTOR (31 downto 0);
    number_V_empty_n : IN STD_LOGIC;
    number_V_read : OUT STD_LOGIC;
    out_V_din : OUT STD_LOGIC_VECTOR (31 downto 0);
    out_V_full_n : IN STD_LOGIC;
    out_V_write : OUT STD_LOGIC );
end;
```

johanpel, 2018-11-29:
This was explored in #88 (filter_hls) and works quite well, I will make it my priority to automate this more through Fletchgen

EraYaN, 2018-12-05:
Might be useful to wrap arrow schema columns in a C style struct. (Like one would in a 

A schema in pyarrow as follows (as example):
```
pyarrow.Field<id: int32>
pyarrow.Field<int1: int32>
pyarrow.Field<int2: int32>
pyarrow.Field<string1: string>
pyarrow.Field<timestamp1: timestamp[ms]>
```
Would convert/generate to 

```cpp
struct schema1 {
    hls::stream<int32_t> &id;
    hls::stream<int32_t> &int1;
    hls::stream<int32_t> &int2;
    hls::stream<char*> &string1; // Fixed length? (char[40])
    hls::stream<uint64_t> &timestamp1;
} ;
```
This would stop function prototypes from growing to huge sizes. With function prototypes as follows, one input per schema file.
```cpp
void increment(schema1 input, schema2 output) {
	/* body */
}
```
or
```cpp
void increment(schema1 &input, schema2 &output) {
	/* body */
}
```

johanpel, 2018-12-10:
An Arrow utf8 string (with an elements-per-cycle configuration of 1) would result in:
hls::stream<int32_t> length
hls::stream<char> values

This way you may decide yourself how you iterate over the strings in your data set.

Higher EPC is possible and I've tested this wrapping four chars in a uint32_t.
Fletcher piggybacks the valid element count onto the values stream.
It's sort of hard to interface this transparently since hls::stream doesn't take structs or unions as a template parameter (as far as I can tell).
So for Vivado HLS it would be more logical to provide a separate stream for the valid element count on the function interface requiring a stream splitter before entering the HLS generated component.

EraYaN, 2018-12-10:
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2016_2/ug902-vivado-high-level-synthesis.pdf page 221
says structs are possible.

>Given a stream specified as hls::stream<T>, the type T may be:
>* Any C++ native data type
>* A Vivado HLS arbitrary precision type (for example, ap_int<>, ap_ufixed<>)
>* A user-defined struct containing either of the above types
>
>**_Note:_** General user-defined classes (or structures) that contain methods (member functions) should not be used as the type (T) for a stream variable.

johanpel, 2019-07-09:
#128 contains the latest progress on this

johanpel, 2021-01-07:
Although this is still something that would be nice to have, it's not critical, so I'm closing this. If anyone wants this, we can reopen it.

closed 2021-01-07 by johanpel