abs-tudelft/fletcher #63: Document MMIO register layout

lars2309, 2018-10-11:
There should be a reference document for the standard / expected MMIO register layout to clarify which implementations are "correct".

Current standard registers (needs to be verified):
- control
- status
- return 0
- return 1
- idx_first
- idx_last
- column 0 address low  (which goes first for list column, offsets or data?)
- column 0 address hi
- column 1 address low
- column 1 address hi
- ...

johanpel, 2018-10-12:
This is a good point.

The flattening of Schema's and arrow::Array buffers should probably also be centralized. Currently this takes place in several tools; the run-time lib and fletchgen. It's perhaps good to set up some common code directory to share stuff between the sub-projects. Also things like logging come to mind.

johanpel, 2018-12-11:
A description is now available here https://github.com/johanpel/fletcher/blob/develop/docs/mmio.md

closed 2018-12-11 by johanpel