abs-tudelft/fletcher #140: Use fileystem library to deal with paths in Cerata and Fletchgen

mbrobbel, 2019-07-05:
Currently the type of paths in Cerata and Fletchgen is `std::string`. This makes handling them not very ergonomic (see #139), which is why I'd like to suggest we use the [filesystem](https://en.cppreference.com/w/cpp/filesystem) library for this. Since we're already using C++17 features, this should not be an issue. It may however be required for some people to update their compilers.

EraYaN, 2019-08-08:
In my fix for #176 this would also be very useful, since now the paths are just joined using `+`. That is not very robust.

johanpel, 2020-07-27:
This is now an issue in Cerata: https://github.com/abs-tudelft/cerata/issues/6

closed 2020-07-27 by johanpel

EraYaN, 2020-07-29:
@johanpel You should be able to move the actual issue between organization projects. (The "Transfer issue" link on the right)