abs-tudelft/fletcher #174: [Python] Queueing RecordBatch results in a segfault for snap platform.

EraYaN, 2019-08-01:
**Describe the bug**
Calling the function `queue_record_batch` with an input batch that results form a `RecordBatchFileReader.get_batch` call, results in a segfault

**To Reproduce**
```py
    parser = argparse.ArgumentParser()
    parser.add_argument("schema_path")
    parser.add_argument("input_path")
    parser.add_argument("output_path")
    args = parser.parse_args()

    schema = pa.read_schema(args.schema_path)
    # Set up a RecordBatch reader and read the RecordBatch.
    reader = pa.RecordBatchFileReader(args.input_path)
    input_batch = reader.get_batch(0)

    output_batch = pa.RecordBatch.from_arrays([pa.array([0] * input_batch.num_rows, pa.uint32())],schema)

    print("Got the batches.")

    platform = pf.Platform("snap", False)                         # Create an interface to an auto-detected FPGA Platform.
    platform.init()                                  # Initialize the Platform.

    print("Initialized platform.")
    context = pf.Context(platform)                   # Create a Context for our data on the Platform.
    print("Created context.")
    context.queue_record_batch(input_batch)                # Queue the RecordBatch to the Context.
    print("Queued record output batch.")
    context.queue_record_batch(output_batch)                # Queue the RecordBatch to the Context.
    print("Queued record output batch.")
    context.enable()                                 # Enable the Context, (potentially transferring the data to FPGA).
```
Run this and the code will fail right after printing: `Created context.`
`schema_path` is the output schema used for the output_path

**Expected behavior**
To continue, just like the same function call in the C++ runtime.

mbrobbel, 2019-08-29:
I'm unable to reproduce the segfault. Can you elaborate a bit on how to reproduce this? Did it work for simulation?

johanpel, 2020-07-27:
Since we cannot reproduce this, I will close it until someone runs into it again.

closed 2020-07-27 by johanpel