abs-tudelft/fletcher #214: Pin all dependencies

mbrobbel, 2019-12-11:
We're still using some unpinned dependencies which may cause build failures due to upstream changes. We should make sure that all dependencies are pinned.

johanpel, 2021-01-07:
With the exception of dependencies on: https://github.com/abs-tudelft/cmake-modules.git
there are currently no unpinned dependencies anymore.
Would you like to pin it on the latest commit there or would you like to create a tag @mbrobbel  ?

mbrobbel, 2021-01-07:
> With the exception of dependencies on: https://github.com/abs-tudelft/cmake-modules.git
> there are currently no unpinned dependencies anymore.
> Would you like to pin it on the latest commit there or would you like to create a tag @mbrobbel ?

Pinning to the lastest commit is fine 👍

johanpel, 2021-01-07:
You know what, let's just make an exception for cmake-modules, it's stable and rarely changes.
Saves us a bunch of commits if it ever does to fix a bug here or there.
Hence I can close this.

closed 2021-01-07 by johanpel