abs-tudelft/fletcher #224: [Fletchgen||C++] Builds with Clang fail since Arrow 0.17 upgrade

johanpel, 2020-04-29:
It seems that upgrading to 0.17 causes Clang to fail building Fletchgen, while GCC succeeds.

Initially I get stuff like:
```
fletcher/common/cpp/src/fletcher/arrow-utils.cc:164:3: error: ignoring return value of function declared with 'warn_unused_result' attribute [-Werror,-Wunused-result]
  fis->Close();
```

In an attempt to fix this it seems that this code in Arrow from arrow/result.h
```
#if __cplusplus >= 201703L
using std::launder;
#else
template <class T>
constexpr T* launder(T* p) noexcept {
  return p;
}
#endif
```

causes additional problems, since std::launder isn't there.

It's probably some discrepancy between GCC and Clang concerning that launder library in C++17, whatever that thing is.

gcc (Ubuntu 8.4.0-1ubuntu1~18.04) 8.4.0
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

mbrobbel, 2020-04-29:
@johanpel With
```cpp
auto status = fis->Close();
```
it compiles for me. (However, this obviously still ignores the status)

johanpel, 2020-07-27:
Seems to work fine now with Clang 10.
I will close this, we can reopen if it poses any issues for anyone else.

closed 2020-07-27 by johanpel